adafruit/Adafruit_nRF52_nrfutil #1: change to python 3

ladyada, 2018-06-22:
@hathach does this have to be python 2 due to a dependancy? if it can be converted to python 3, then that would be good for long-term supportability!

hathach, 2018-06-23:
yeah, we should either move to python 3 or going with a C port. The nrfutil mostly use serial port which should be easy to access from either language ( I am not really good with python :( ) 

https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/65

hathach, 2018-06-24:
Ah thanks, I will try, but my Python knowledge is quite limited. Almost forgot everything every time I  got to work with Python :(

ladyada, 2018-06-25:
not a problem. it should be all done so if you can test it and let me know if anything doesnt work that is best. i only tried dfu upload and genpkg!

ladyada, 2018-08-12:
@hathach

closed 2018-08-12 by ladyada