adafruit/Adafruit_nRF52_nrfutil #7: blocked by McAfee anti-virus

hathach, 2018-09-18:
This may have no fix, but just create issue for reminder. A user got adafruit-nrfutil.exe blocked by above antivirus program

https://forums.adafruit.com/viewtopic.php?f=57&t=136142&p=697388&hilit=nrf52#p697388

ladyada, 2018-09-18:
@hathach we can submit it and they will remove the false positive
https://service.mcafee.com/webcenter/portal/oracle/webcenter/page/scopedMD/s55728c97_466d_4ddb_952d_05484ea932c6/Page29.jspx?wc.contextURL=%2Fspaces%2Fcp&locale=en-US&articleId=TS102053&_afrLoop=2171358329332895&leftWidth=0%25&showFooter=false&showHeader=false&rightWidth=0%25&centerWidth=100%25#!%40%40%3FshowFooter%3Dfalse%26locale%3Den-US%26_afrLoop%3D2171358329332895%26articleId%3DTS102053%26leftWidth%3D0%2525%26showHeader%3Dfalse%26wc.contextURL%3D%252Fspaces%252Fcp%26rightWidth%3D0%2525%26centerWidth%3D100%2525%26_adf.ctrl-state%3D15g9m9xs68_9

can you please email them the file - i can also try, i've done this a few times before, its common :)

[image]

hathach, 2018-09-18:
Wow, I don't know we can do this. I could send them a email, but It is better if you could do it. Since my adafruit email is received only. Sending them the request with personal email won't really work I guess.

ladyada, 2018-09-18:
it doesnt matter the email address, they will analyze it on request. please try!

hathach, 2018-09-18:
Ok, I will do it

hathach, 2018-09-25:
user still haven't responded with McAfee version and Engine that requires by McAffe yet. Just ask him on the topic, maybe he forgot to stick the notify by email.

[image]

hathach, 2019-02-11:
response from AV

```
McAfee Labs - Beaverton                                                                
Current Scan Engine Version:6000.8403                                                  
Current DAT Version:9146.0000                                                          
Thank you for your submission.                                                         

Analysis ID: 10746694

File Name            Findings                       Detection                    Type         Extra
--------------------|------------------------------|----------------------------|------------|-----
adafruit-nrfutil.exe|inconclusive                  |                            |            |no   

inconclusive [adafruit-nrfutil.exe]                                                                

   Automated analysis was not able to determine that this file is malware. This file is   
being sent for further processing and the DAT files will potentially be updated if     
detection of this sample is warranted.             
```

closed 2019-02-11 by hathach