adafruit/Adafruit_nRF52_nrfutil #8: Prematurely exit after programming in serial DFU

martin-intex, 2019-01-19:
Under certain circumstances adafruit-nrfutil exits prematurely before the flash is fully written when using the serial DFU mode (e.g. in the Adafruit Feather nRF52 Bluefruit). If a reset is issued shortly thereafter (e.g. by an opened serial monitor of the Arduino GUI)  the flash operation is not finished and the flash content becomes corrupted. The problem occurred in the following configuration:

Windows 10 OS running xubuntu 18.04  inside VirtualBox 6.0 with Oracle VirtualBox Extension Pack and emulated USB3.0 (xHCI) peripheral. 

Use the attached Arduino project to reproduce the test. SPI.h must be included as well as enabling debug level 2.

martin-intex, 2019-01-19:
Attaching .ino files seems to be not allowed. So here is the code:

---------------------------:<--------------------------------
/*Use Debugging==Level 2 in Tools/Debug Mode*/
//Including SPI header is important - most probably it increases .text and/or .bss size which
//will trigger the problem
#include <SPI.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Init XXX ....");
}

void loop() { 
  //This fails 
 Serial.printf("Lalala3\n\r");

 //This works
 //Serial.println("Lalala2");

 delay(1000);
}
---------------------------------:<-----------------------------------

hathach, 2019-01-20:
@martin-intex thank you very much for openning this issue. Your setup is a bit advanced, can you test uploading with native OS without virtualbox invovled, so that we could easily reproduce it. Once reproduced, we will further troubleshoot it.

martin-intex, 2019-01-20:
I tried with native Linux as well as Windows but unfortunately could not reproduce the issue there. I however can provide some debugging findings. The problem is that the "magic" bytes at 0x7f000 ... 0x7f010 are zero in the non-working version but set to the correct value in the working one. The remaining flash is identical between the working and non-working version.

hathach, 2019-01-20:
If it is not reproducible under native setup. I don't see the need to "fix" this anytime soon. Since we can't verify if the fix when troubleshooting. Though we will be happy to merge if submit a PR for this.

hathach, 2020-09-30:
closed due to lack of activities.

closed 2020-09-30 by hathach