adafruit/Adafruit_nRF52_nrfutil #9: Simultaneous Python 2 and 3 support

jeremypoulter, 2019-01-22:
Would it be possible to support both Python 2 and 3? It is still common to only have Python 2 installed, it would be nice to not have to install Python 3 just for this. 

For background I am trying to get my Platform IO port of the Adafruit nRF52 BSP building in [Travis CI](https://travis-ci.org/jeremypoulter/platform-nordicnrf52/), that checks both 2.7 and 3.5 (Platform IO if Python based).

ladyada, 2019-01-22:
python2 is deprecated now - and officially unsupported after this year - we have to move forward, its been 10 years! :)

closed 2019-01-22 by ladyada

jeremypoulter, 2019-01-23:
Fair enough, though I would ask ;-) I have fixed the issues on the build (by installing Python3) so hopefully the Platform IO maintainers have a similar attitude to Python.