adafruit/Adafruit_nRF52_nrfutil #10: Fatal Python error when running adafruit-nrfutil

boriswtam, 2019-02-20:
I followed the instructions to install adafruit-nrfutil from source, but I have been getting the error: 

Fatal Python error: Py_Initialize: unable to load the file system codec
LookupError: unknown encoding: utf-8

Current thread 0x00007fffed06b3c0 (most recent call first):
Abort trap: 6
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...82 completed.

I'm currently running MacOS 10.12 and am trying to program my NRF52840 Mini

hathach, 2019-02-20:
For macos, please use either prebuilt binary or install it from pypi package.

hathach, 2020-09-30:
closed due to lack of activities.

closed 2020-09-30 by hathach