adafruit/Adafruit_nRF52_nrfutil #11: Flashing a SoftDevice profile (S340) abruptly exits

judepereira, 2019-07-10:
I got the S340 SoftDevice hex file from thisisant.com. I'm trying to flash it using the following commands:
```
$ adafruit-nrfutil dfu genpkg --sd-req 0xFFFE --dev-type 0x0052 --softdevice ../ext/ANT_s340_nrf52840_6.1.1.hex _build/dfu-s340-package.zip
$ adafruit-nrfutil --verbose dfu serial --package _build/dfu-s340-package.zip -p /dev/tty.usbmodem14301 -b 115200 --singlebank --touch 1200
```

The output of the dfu serial command is here:
```
Upgrading target on /dev/tty.usbmodem14301 with DFU package /Users/jude/developer/em/bia/src/_build/dfu-s340-package.zip. Flow control is disabled, Single bank, Touch 1200
Touched serial port /dev/tty.usbmodem14301
Opened serial port /dev/tty.usbmodem14301
Starting DFU upgrade of type 1, SoftDevice size: 190272, bootloader size: 0, application size: 0
Sending DFU start packet
Sending DFU init packet
Sending firmware file

Failed to upgrade target. Error is: write failed: [Errno 6] Device not configured
Traceback (most recent call last):
  File "/Users/jude/Library/Python/3.7/lib/python/site-packages/serial/serialposix.py", line 537, in write
    n = os.write(self.fd, d)
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/jude/Library/Python/3.7/lib/python/site-packages/nordicsemi/__main__.py", line 294, in serial
    dfu.dfu_send_images()
  File "/Users/jude/Library/Python/3.7/lib/python/site-packages/nordicsemi/dfu/dfu.py", line 229, in dfu_send_images
    self._dfu_send_image(HexType.SOFTDEVICE, self.manifest.softdevice)
  File "/Users/jude/Library/Python/3.7/lib/python/site-packages/nordicsemi/dfu/dfu.py", line 206, in _dfu_send_image
    self.dfu_transport.send_firmware(firmware)
  File "/Users/jude/Library/Python/3.7/lib/python/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 213, in send_firmware
    self.send_packet(pkt)
  File "/Users/jude/Library/Python/3.7/lib/python/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 241, in send_packet
    self.serial_port.write(bytearray(pkt.data))
  File "/Users/jude/Library/Python/3.7/lib/python/site-packages/serial/serialposix.py", line 571, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.
```

What appears to happen is that after some bytes are transferred, it seems to stop abruptly. 

To debug what's happening, I changed the logging level from INFO to DEBUG (where it prints a whole bunch of data), but couldn't gather anything from it (prints too much data bytes).

My board is the SparkFun NRF52840 Mini (the boot loader is based on the Adafruit one).

Furthermore, I have a few questions:
1. If a SoftDevice is flashed, how does the boot loader know which address does the application start from? Is this only because of the linker script where the flash origin and size is mentioned?
2. Since the NRF52840 supports UF2, the contents of INFO_UF2.TXT are as follows:
```
UF2 Bootloader 1.00
Model: SparkFun nRF52840 Mini
Board-ID: SparkFun nRF52840 Mini
Bootloader: s140 6.1.1 r0
Date: Nov 26 2018
```

This indicates that the SoftDevice info comes from the boot loader. Does this mean that the boot loader *and* the SoftDevice are coupled together?

ladyada, 2019-07-10:
hi are you using "SparkFun nRF52840 Mini" hardware?

judepereira, 2019-07-10:
Yes, I'm usng that hardware.

ladyada, 2019-07-10:
Hiya, you may want to ping @loricrotser @jimblom https://learn.sparkfun.com/tutorials/sparkfun-pro-nrf52840-mini-hookup-guide/all to get assistance

judepereira, 2019-07-10:
Thanks, I did leave a comment on that very page. Also, SparkFun's forum seems to be down :(

judepereira, 2019-07-10:
@ladyada any chance you can describe how the boot loader works with the SoftDevice? Or any links you can point me to?

ladyada, 2019-07-10:
Hi please check out https://forum.sparkfun.com/viewtopic.php?t=50272&p=205738#p205738

judepereira, 2019-07-10:
Looks like the SparkFun forums are back online. Thanks! I'll post this there as well. Feel free to close this issue here if you think it's irrelevant for this repo.

ladyada, 2019-07-10:
ok!

closed 2019-07-10 by ladyada

drewvigne, 2020-05-05:
I recently successfully flashed the s340 to my adafruit nrf52840. If you still need a resolution I would be happy to share my method.

HristoMarkow, 2022-12-05:
> I recently successfully flashed the s340 to my adafruit nrf52840. If you still need a resolution I would be happy to share my method.

I'm new in this, but Seeeduino XIAO can be flashed same way, right? It is based on nRF52840