adafruit/Adafruit_nRF52_nrfutil #12: adafruit-nrfutil keys --gen-key private.pem fails

zzasada, 2019-10-09:
Running the command adafruit-nrfutil keys --gen-key private.pem fails with error

`certificates\vault>adafruit-nrfutil keys --gen-key private.pem
Traceback (most recent call last):
  File "c:\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\zzasa\AppData\Roaming\Python\Python37\Scripts\adafruit-nrfutil.exe\__main__.py", line 9, in <module>
  File "c:\python37\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\python37\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\python37\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python37\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python37\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\zzasa\AppData\Roaming\Python\Python37\site-packages\nordicsemi\__main__.py", line 126, in keys
    signer.gen_key(key_file)
  File "C:\Users\zzasa\AppData\Roaming\Python\Python37\site-packages\nordicsemi\dfu\signing.py", line 35, in gen_key
    sk_file.write(self.sk.to_pem())
TypeError: write() argument must be str, not bytes`

Running version
adafruit-nrfutil version 0.5.3.post12

ladyada, 2019-10-09:
ahh this is probably left over frmo our py2 to py3 port. could you try to fix and submit a patch? it should just a `b"abcde".decode("utf-8") ` type thing

JuggernautPanda, 2020-07-15:
Hi, this error still persists and has not been fixed.

hathach, 2020-07-16:
This is not supported, adafruit-nrfutil doesn't support secured DFU. Please use the original nrfutil from Nordic

closed 2020-07-16 by hathach