adafruit/Adafruit_nRF52_nrfutil #17: ModuleNotFoundError: No module named 'nordicsemi.__main__' - issue with latest pip install

JuggernautPanda, 2020-07-15:
Hello. I recently updated the package and I am getting this error. It was working fine earlier.

Now I am unable to use the tool.

JuggernautPanda, 2020-07-15:
adafruit-nrfutil version
Traceback (most recent call last):
  File "/usr/local/bin/adafruit-nrfutil", line 5, in <module>
    from nordicsemi.__main__ import cli
ModuleNotFoundError: No module named 'nordicsemi.__main__'

JuggernautPanda, 2020-07-15:
OS: Ubuntu 18.04

JuggernautPanda, 2020-07-15:
I got the issue. I tried to install nrfutil package as well.
 
This was causing the issue. I uninstalled both completely for both python3 and python2.7
sudo pip3 uninstall adafruit-nrfutil
sudo pip3 uninstall nrfutil
sudo pip uninstall adafruit-nrfutil
sudo pip uninstall nrfutil

Then I closed the terminal to open a new session and then reinstalled it by using
sudo pip3 install adafruit-nrfutil

closed 2020-07-15 by JuggernautPanda

glorl, 2020-12-26:
Thank @JuggernautPanda, this solved the same problem for me

artgodwin, 2022-05-26:
I tried to follow these instructions, but pip told me it was out of date and to update it.

After I updated it, every attempt to use pip fails with the error

1, in main
    from pip._internal.utils.entrypoints import _wrapper
  File "~/.local/lib/python3.5/site-packages/pip/_internal/utils/entrypoints.py", line 12
    f"pip{sys.version_info.major}",
                                 ^

hathach, 2022-05-26:
@artgodwin can you provide more info on your PC: device, OS version. Your python is pretty old 3.5, I think you probably need to update to later python version, or just use the old nrfutil that worked previously.

dhalbert, 2022-05-27:
@hathach we took this to discord

hathach, 2022-05-27:
thanks @dhalbert

artgodwin, 2023-05-10:
> was there any solution?

I got it to work by running the whole Arduino IDE inside a python ve. I am
running the python 3 shipped with Debian. This obviously isn't a very good
solution and I think using python for build utilities is fairly risky : you
need to be very conservative unless you ship the appropriate version with
the BSP.

Ultimately I moved to the Nordic SDK instead of Arduino.