adafruit/Adafruit_nRF52_nrfutil #22: Error trying to upgrade bootloader on Adafruit Feather nRF52840 Express

dbhaig, 2020-12-09:
I am following the procedure outlined here: https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather?view=all
and trying to update the bootloader using adafruit-nrfutil (nrfutil version 6.1.0)

When I execute the command:

adafruit-nrfutil  dfu serial --package /home/don/Downloads/feather_nrf52840_express_bootloader-0.3.2_s140_6.1.1.zip -p /dev/ttyACM0 -b 115200 --singlebank --touch 1200

I get: Error: no such option: --singlebank, Error: no such option: --touch

When I remove these options and run the command below, the application crashes:

./adafruit-nrfutil  dfu serial --package /home/don/Downloads/feather_nrf52840_express_bootloader-0.3.2_s140_6.1.1.zip -p /dev/ttyACM0 -b 115200
Traceback (most recent call last):
  File "./adafruit-nrfutil", line 11, in <module>
    load_entry_point('adafruit-nrfutil==0.5.3.post13', 'console_scripts', 'adafruit-nrfutil')()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/don/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 1055, in serial
    do_serial(package, port, connect_delay, flow_control, packet_receipt_notification, baud_rate, serial_number, True,
  File "/home/don/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 964, in do_serial
    dfu = Dfu(zip_file_path = package, dfu_transport = serial_backend, connect_delay = connect_delay)
  File "/home/don/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 69, in __init__
    self.manifest           = Package.unpack_package(zip_file_path, self.unpacked_zip_path)
  File "/home/don/.local/lib/python3.8/site-packages/nordicsemi/dfu/package.py", line 654, in unpack_package
    return Manifest.from_json(_json)
  File "/home/don/.local/lib/python3.8/site-packages/nordicsemi/dfu/manifest.py", line 200, in from_json
    return Manifest(**kwargs['manifest'])
TypeError: __init__() got an unexpected keyword argument 'dfu_version'

I have tried installing adafruit-nrfutil using pip3 as well as from source.  
I am running Ubuntu 20.04.
Is there something that I'm missing?
Thanks

hathach, 2020-12-10:
what is your version command return and how did you install the tool

```
$ adafruit-nrfutil version
adafruit-nrfutil version 0.5.3.post13
```

dbhaig, 2020-12-10:
$ adafruit-nrfutil version
nrfutil version 6.1.0

First I installed the tool using: 
$ pip3 install --user adafruit-nrfutil

When that didn't work, I cloned the repository:
$ git clone https://github.com/adafruit/Adafruit_nRF52_nrfutil.git
$ cd Adafruit_nRF52_nrfutil
$ pip3 install -r --user requirements.txt
$ python3 setup.py install

I got the same error both times.

nrfutil is also installed (python -m pip install nrfutil) if that makes a difference.
$ nrfutil version
nrfutil version 6.1.0

dbhaig, 2020-12-10:
So having nrfutil installed was the source of the problem. Uninstalling both packages and reinstalling adafruit-nrfutil solved it.

Thanks for taking the time to help me sort this out.

closed 2020-12-10 by dbhaig

hathach, 2020-12-11:
@dhalbert installed both Nordic nrfutil & adafruit-nrfutil seems to interfere with each other, maybe they share some "ID" somewhere, not an urgent issue though :)