adafruit/Adafruit_nRF52_nrfutil #24: New build with Windows and macOS binaries

deadprogram, 2021-02-07:
Would it be possible to please publish binaries for Windows and macOS of the latest release. Thank you!

hathach, 2021-02-08:
@dhalbert I think we previously use travis ci for building binaries, but it isn't migrated to github action for this repo just yet.

vmocanu, 2021-02-25:
Adafruit nRF Feather bootloader instructions refers to this page for downloading the windows exe, but there seems to be none here. Any advice on how to proceed? I'm stuck atm.
Thanks!

hathach, 2021-02-25:
you can download the previous release here https://github.com/adafruit/Adafruit_nRF52_nrfutil/releases/tag/%24(APPVEYOR_REPO_TAG_NAME) 

For windows, it is the same as the latest fix issue on Linux only. We migrated from travis to github ci and didn't get the later to build the release binary yet. WIll do later on when having time.

vmocanu, 2021-02-25:
> you can download the previous release here 0.5.3post10 https://github.com/adafruit/Adafruit_nRF52_nrfutil/releases/tag/%24(APPVEYOR_REPO_TAG_NAME)
> 
> For windows, it is the same as the latest fix issue on Linux only. We migrated from travis to github ci and didn't get the later to build the release binary yet. WIll do later on when having time.
> 
> Note: you can also install it with pip3 as well on windows.

Thank you for the hint!
Indeed I finally figured out how to build and install it with pip. All fine now.

mej558, 2021-03-11:
"you can download the previous release here 0.5.3post10 https://github.com/adafruit/Adafruit_nRF52_nrfutil/releases/tag/%24(APPVEYOR_REPO_TAG_NAME)"

Is that version meant to run on Windows? This is the error I received... I don't know Python on pip3. I tried adding those folders manually with no success. I would love to just have this just work right... Can you upload something working like the Adafruit directions suggest it will? Thanks mike

C:\Users\mike\Downloads>adafruit-nrfutil.exe --verbose dfu serial --package feather_nrf52840_express_bootloader-0.4.1.zip --port COM5 -b 115200 --singlebank --touch 1200
Traceback (most recent call last):
  File "nordicsemi\__main__.py", line 315, in <module>
  File "site-packages\click\core.py", line 764, in __call__
  File "site-packages\click\core.py", line 717, in main
  File "site-packages\click\core.py", line 1137, in invoke
  File "site-packages\click\core.py", line 1137, in invoke
  File "site-packages\click\core.py", line 956, in invoke
  File "site-packages\click\core.py", line 555, in invoke
  File "nordicsemi\__main__.py", line 288, in serial
  File "nordicsemi\dfu\dfu.py", line 66, in __init__
  File "nordicsemi\dfu\package.py", line 364, in unpack_package
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\mike\\AppData\\Local\\Temp\\nrf_dfu_oagbyp40\\unpacked_zip\\manifest.json'
[2892] Failed to execute script __main__

hathach, 2021-03-14:
@mej558 yes, it should work on windows 7, 8 and 10. Can you tell me what is your windows version ?

tadakado, 2021-04-25:
Hi, I have similar error message on mac0S (11.2.3) with precompiled one as well as "pip3 install --user adafruit-nrfutil" (pypi) and "pip3 install --user -r requirements.txt" (source).
Below is the message from precompiled one.

% ./adafruit-nrfutil-macos --verbose dfu serial --package feather_nrf52840_express_bootloader-0.5.0.zip -p /dev/cu.usbmodem1101 -b 115200 --singlebank --touch 1200

Traceback (most recent call last):
  File "nordicsemi/__main__.py", line 315, in <module>
  File "click/core.py", line 764, in __call__
  File "click/core.py", line 717, in main
  File "click/core.py", line 1137, in invoke
  File "click/core.py", line 1137, in invoke
  File "click/core.py", line 956, in invoke
  File "click/core.py", line 555, in invoke
  File "nordicsemi/__main__.py", line 288, in serial
  File "nordicsemi/dfu/dfu.py", line 66, in __init__
  File "nordicsemi/dfu/package.py", line 364, in unpack_package
FileNotFoundError: [Errno 2] No such file or directory: '/var/folders/x7/xwc5rvj97r76wftxwhzt36h00000gn/T/nrf_dfu_0qrjosh9/unpacked_zip/manifest.json'
[24779] Failed to execute script __main__

hathach, 2021-05-18:
I have just create a new binary based on new release of post16 
https://github.com/adafruit/Adafruit_nRF52_nrfutil/releases/tag/0.5.3.post16

Please try to see if that works out for you, here is the machine I used to create the binaries
- windows 10 , python 3.9
- macOS 11.1, python 3.8.5

makermelissa, 2021-05-18:
GitHub Actions was down. Now that it's back up, I just re-ran the binary release actions.

closed 2021-05-25 by hathach

tadakado, 2021-05-27:
The boot loader zip file is in the distributed zip file, so I needed to extract it.
I successfully flashed the firmware with correct command:

% ./adafruit-nrfutil-macos --verbose dfu serial --package feather_nrf52840_express_bootloader-0.5.0_s140_6.1.1.zip -p /dev/cu.usbmodem1101 -b 115200 --singlebank --touch 1200

hathach, 2021-05-27:
thanks for confirmation, I am glad that help