adafruit/Adafruit_nRF52_nrfutil #28: adafruit-nrfutil-0.5.3.post16-win.zip file contains Trojan:Win32/Sabsik.FL.A!ml

sumancvb, 2021-05-19:
Latest release:  0.5.3.post16

file: adafruit-nrfutil-0.5.3.post16-win.zip

detected as Trojan:Win32/Sabsik.FL.A!ml in windows 10

[image]

hathach, 2021-05-19:
hmm, this file is previously false alarm by mcafee as well. 
- Is the version included by bsp core 0.5.3.post12 is detected positive as well ?
- can you try to install it from `pip install adafruit-nrfutil` to see if that works

hathach, 2021-05-19:
PS: can you try to download the win.zip version from this test release to see if that is any better https://github.com/hathach/Adafruit_nRF52_nrfutil/releases/tag/0.5.3.test18

closed 2021-05-20 by sumancvb

hathach, 2021-05-20:
thanks for confirmation, it still bug me with the false positive. Since I normally bundle exe for bsp release. I will try to find a better way somehow.