adafruit/Adafruit_nRF52_nrfutil #29: Windows 10 install of adafruit-nrfutil fails (2021)

gfmoore, 2021-05-23:
Hi, I've been struggling to install the Adafruit-nrfutil following the pip instructions. (pip doesn't seem to function now on Python 3.9.5 have to use python -m pip apparently?

Anyway I did  
```
python -m pip install --user adafruit-nrfutil 
``` 
and got error
```
failed to create process.
 ```

Thankfully in the previous issue a windows executable was created and this worked, but still thought I'd raise it.

hathach, 2021-05-24:
There seems to be an issue with your python installation, others including me has no problem install it from pip with windows.
1. First of all, can you tell us what is your windows version 7 or 10 and its build version as well.
2. Please try to uninstall and reinstall python https://www.python.org/downloads/ here, make sure you select Add pip/python to your PATH if any.
3. try to install it with `pip install adafruit-nrfutil` to see if that work out. If not please paste your output here.

gfmoore, 2021-05-24:
Hi, thanks. You were right, the Python installation was a right mess - probably from trying to use the official installer and MS rubbish. Once I'd cleared away Python 3.7 and stopped the Windows store and fixed my paths I was able to get the thing working and now I was able to use the the thing to install the bootloader on my nRF52840 MDK device and get blinky working. 

I still have to use py -m pip ... (I assume I could create an alias for that but for now it will do.) Thanks for replying nd offering the advice it is very much appreciated. I was starting to lose my mind...

Closed

closed 2021-05-24 by gfmoore

hathach, 2021-05-24:
I am glad you got it worked out, I am sure the pip thing is probably only the PATH issue. There is probably more people with the same issue, just google around if it become too annoying.