adafruit/Adafruit_nRF52_nrfutil #31: Pitaya Go port not open error

rcmz, 2021-08-19:
Hello I'm trying to install CircuitPython on the Pitaya Go but I get a port not open error:
```
$ adafruit-nrfutil dfu serial -pkg pitaya_go_bootloader-0.6.1_s140_6.1.1.zip -p /dev/ttyACM0 -b 115200
                    
Upgrading target on /dev/ttyACM0 with DFU package /home/rcmz/Downloads/pitaya_go_bootloader-0.6.1_s140_6.1.1.zip. Flow control is disabled, Dual bank, Touch disabled
Timed out waiting for acknowledgement from device.

Failed to upgrade target. Error is: Attempting to use a port that is not open
Traceback (most recent call last):
  File "/home/rcmz/.local/lib/python3.9/site-packages/nordicsemi/__main__.py", line 296, in serial
    dfu.dfu_send_images()
  File "/home/rcmz/.local/lib/python3.9/site-packages/nordicsemi/dfu/dfu.py", line 226, in dfu_send_images
    self._dfu_send_image(HexType.SD_BL, self.manifest.softdevice_bootloader)
  File "/home/rcmz/.local/lib/python3.9/site-packages/nordicsemi/dfu/dfu.py", line 203, in _dfu_send_image
    self.dfu_transport.send_init_packet(init_packet)
  File "/home/rcmz/.local/lib/python3.9/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 155, in send_init_packet
    self.send_packet(packet)
  File "/home/rcmz/.local/lib/python3.9/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 241, in send_packet
    self.serial_port.write(bytearray(pkt.data))
  File "/usr/lib/python3.9/site-packages/serial/serialposix.py", line 615, in write
    raise PortNotOpenError()
serial.serialutil.PortNotOpenError: Attempting to use a port that is not open

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.
```

hathach, 2021-08-19:
Which bootloader version your board is running ? Double reset then paste your info.txt here. Also You should use uf2 to install circuitpyhton instead

rcmz, 2021-08-19:
How can I know what version my board is running and how can I extract info.txt ? I'm trying to install the uf2 bootloader (https://github.com/adafruit/Adafruit_nRF52_Bootloader) in order to then install circuitpython, because the pitaya go did not show up as a uf2 drive by default

hathach, 2021-08-19:
If so, you will probably need external programmer like jlink to flash this uf2 bootloader. Try to post this question to the vendor support forum, they make the board and could help you out. Closed sine it is out of scope of this repo

closed 2021-08-19 by hathach