adafruit/Adafruit_nRF52_nrfutil #35: Crashing while flashing

JesseSwildens, 2021-12-06:
### Operating System

Windows 10

### IDE version

nrfutil-0.5.3.post16/17

### Board

PCA10059

### Version

nrfutil-0.5.3.post16/17

### What happened ?

When starting the flashing program with command: adafruit-nrfutil.exe --verbose dfu serial --package pca10059_bootloader-0.6.1.zip --port COM6 -b 115200 --singlebank --touch 1200 . The program crashes while using.

### How to reproduce ?

downloaded the latest versions and normal use

### Debug Log

Traceback (most recent call last):
  File "__main__.py", line 317, in <module>
  File "click\core.py", line 1134, in __call__
  File "click\core.py", line 1059, in main
  File "click\core.py", line 1665, in invoke
  File "click\core.py", line 1665, in invoke
  File "click\core.py", line 1401, in invoke
  File "click\core.py", line 767, in invoke
  File "__main__.py", line 290, in serial
  File "dfu\dfu.py", line 66, in __init__
  File "dfu\package.py", line 364, in unpack_package
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\JESSES~1\\AppData\\Local\\Temp\\nrf_dfu_djy9e8ah\\unpacked_zip\\manifest.json'
[12824] Failed to execute script __main__

hathach, 2021-12-06:
the zip is not correct one, you need to unzip that first and then use the 
- pca10056_bootloader-0.6.2_s140_6.1.1.zip instead

It seems to cause a bit of confusion, maybe I will change to upload zip/hex/uf2 individually instead of packing them together in the next release.

JesseSwildens, 2021-12-06:
Thank you for the suggestion. That was indeed the issue.

closed 2021-12-06 by JesseSwildens