adafruit/Adafruit_nRF52_nrfutil #36: Crowdstrike Falcon Sensor Error

rcrawfo1-brown, 2021-12-07:
### Operating System

Windows 10

### IDE version

1.8.16

### Board

Adafruit Feather nRF52832

### Version

1.20.0

### What happened ?

Enterprise Crowdstrike Falcon Sensor blocks adafruit-nrfutil.exe from running, labeling it as Malware

### How to reproduce ?

1. Install Crowdstrike for home 6.X 
2. Download Arduino 
3. Follow Feather board guide https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide?view=all
4. Attempt to verify Beacon 
5. Compile fails because of CrowdStrike

### Screenshots

[image]

hathach, 2021-12-08:
sigh, I have no idea why AV would block this util. If I would, I could update the code to go around that. Or do you know there is any way for us to submit an `false positive` report for crowdstrike to examine the exe ? Anyways, would you mind trying follow walkaround
1. add adafruit-nrfutil to exception list
2. try to install via pip package with `pip3 install --user adafruit-nrfutil` then locate the exe file and overwrite the one that got blocked.
3. compile from source by clone this repo then `python3 setup.py install`

both 2 and 3 are covered in the readme.

rcrawfo1-brown, 2021-12-10:
Yeah I've got no ideal either. I tried to find a place to submit false positives. There doesn't seem to be an easy place to do that and since I'm using an enterprise software the crowdstrike IT won't talk to me, they'll only talk to the Instution's IT services. So it will take a second. 
So I've pip installed the adafruit-nrfutil but I'm not sure how to overwrite the file. I've located the folder for the exe and the folder for the pip installed adafruit-nrfutil library, but I'm not sure what I should do next or where in the README that is (I've been looking at the Linux pip install instructions). Not super skilled with arduino development so I appreciate the help.

hathach, 2021-12-13:
@rcrawfo1-brown that is a bit of pain, you could manually change the path to the adfaruit-nrfutil.exe by editting the platform.txt

- https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/platform.txt#L137

by changing it to simply `adafruit-nrfutil.exe` since the pip package already add it into the PATH (python3.x/script) in your machine.

```
tools.nrfutil.cmd.windows=adafruit-nrfutil.exe
```

rcrawfo1-brown, 2021-12-14:
hi, so that worked to get the beacon code to compile. However when trying to run the code I get the following issue in the first picture. I assume this is a bootloader error since the documentation says you have to update the bootloader, however when trying to update the bootloader I ran into the same AV issue. My menu is also slightly different than the one in the documentation, saying softdevice instead of bootloader in the tool menu.
[image]
[image]
[image]

hathach, 2021-12-15:
for burning bootloader, you need to update the programmer in similar way
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/programmers.txt#L26

rcrawfo1-brown, 2021-12-20:
Awesome! Seems like that worked! Beacon uploaded successfully. Thank you for your help, I have contacted my IT department to hopefully send a false positive report to Crowdstrike but we'll see.

hathach, 2021-12-21:
thank you for reporting this to IT department. Meanwhile I will try to see if we could detect the tool installed from pip package and use it instead of manual editing.

closed 2022-01-12 by rcrawfo1-brown