adafruit/Adafruit_nRF52_nrfutil #37: genpkg is not found

Gerriko, 2022-04-28:
### Operating System

Linux

### IDE version

Arduino 1.8.16

### Board

nRF52840-DK

### Version

latest

### What happened ?

When trying to upload to my nRF52840 board, Arduino IDE indicates an error in the console `Error: No such command 'genpkg'`

Searching through the source code, I see this function/command is used in __main__.py (line 186) to generate the nRF zip package.

Checking through the latest Nordic nrfutil documention they use: `nrfutil pkg generate...` command to do similar.

### How to reproduce ?

Select an example
Compile and upload.
Error will be generated.

### Debug Log

Usage: adafruit-nrfutil dfu [OPTIONS] COMMAND [ARGS]...
Try "adafruit-nrfutil dfu --help" for help.

Error: No such command "genpkg".
exit status 2
/home/nnnn/arduino-1.8.16/arduino-builder returned 2
Error compiling for board Nordic nRF52840 DK.

hathach, 2022-04-28:
how did you install adafruit-nrfutil and what is its version with `adafruit-nrfutil version`

hathach, 2022-04-28:
The arduino package does not include binary for Linux (only windows and macos). For Linux it is normally installed via pip package. This is rather different from the nrfutil from Nordic, make sure you remove all other and try to re-install it again. mine is post16

```
$ pip3 install adafruit-nrfutil
$ adafruit-nrfutil version
adafruit-nrfutil version 0.5.3.post16
```

Gerriko, 2022-04-30:
I used `pip3 uninstall adafruit-nrfutil` to remove it first. Then I tried to install again using `pip3 install --user adafruit-nrfutil` but it would not let me install as it detected other files... maybe because I left our the --user flag. Who knows.

So I did a `pip3 install --user --force-reinstall adafruit-nrfutil` which worked.
The version I have is as above `adafruit-nrfutil version 0.5.3.post16`

It now works.

Gerriko, 2022-04-30:
It looks like what I did got rid of the original Nordic nrfutil program, which I use for other things as when I tried to use nordic's version then it used Adafruit's version. When I forced a reinstall of the Nordic version then it would not use the Adafruit version.

So, if I type in `adafruit-nrfutil` it actually uses Nordic's nrfutil program. Hence I think this is what is causing the problem.

Gerriko, 2022-05-01:
Note that I got the Nordic nrfutil utility program to work with Arduino by adding the following in the relevant Arduino platforms.txt file.

```
## Create dfu package zip file
recipe.objcopy.zip.pattern="{tools.nrfutil.cmd}"  pkg generate --hw-version 0x34 --sd-req 0x00 --application "{build.path}/{build.project_name}.hex" --application-version 1 "{build.path}/{build.project_name}.zip"

```

hathach, 2022-07-01:
glad you figured this out, I guess we could close this now.

closed 2022-07-01 by hathach