adafruit/Adafruit_nRF52_nrfutil #41: License clarification

pehjota, 2023-09-30:
Hi,

Could someone please clarify which license applies to Nordic's copyrights?  The original project [uses](https://github.com/NordicSemiconductor/pc-nrfutil/commit/48b618967f334743b5937d1283fbf8837c4337e7) a 5-clause [license](https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/LICENSE), which includes prohibitions against use with non-Nordic processors and reverse engineering of binaries.  But in the ["first commit"](https://github.com/adafruit/Adafruit_nRF52_nrfutil/commit/07b43832ee53a4a248c30f5a3014e2632d8aeb88) (i.e. there's no history showing the original license being changed to this one) to this project, @microbuilder added files from the original project with clauses 4 and 5 inexplicably dropped from the [license](https://github.com/adafruit/Adafruit_nRF52_nrfutil/blob/master/license.txt).  Notably though, this project's [nordicsemi/dfu/signing.py](https://github.com/adafruit/Adafruit_nRF52_nrfutil/blob/master/nordicsemi/dfu/signing.py) still refers to a "non-transferable" "NORDIC SEMICONDUCTOR STANDARD SOFTWARE LICENSE AGREEMENT", which as far as I can tell by a Web search is the 5-clause license.

Did the code actually come from Nordic elsewhere with the 3-clause license, did @microbuilder / @adafruit get permission from Nordic to drop clauses 4 and 5, or is the license of this project wrong?

I ask because I've started to package this utility to hopefully be uploaded into Debian, but I know Debian will reject the 5-clause license as non-free.

microbuilder, 2023-09-30:
This was 6 years ago, and I have exactly zero recollection of anything here, but I know I wouldn't have randomly dropped anything from a license. 😜 I suspect I cloned whatever was available at the time, and since it can only be used with Nordic parts complies with the Nordic only clause. 

Indicating that it was BSD3 in GitHub for the repo settings is a mistake, though, and should have probably just been left blank. Perhaps that is where the 3 clause file in root came from?

Sorry for the confusion, but I really have zero recollection of the fork process that happened here a lifetime and a couple companies ago, but I can't imagine a universe where I'm modifying any existing license ever myself.

microbuilder, 2023-09-30:
It looks to me in the first link in your comment ("use") that Nordic itself changed their license in 2016 in that commit from Carles, and we made our fork before those changes (2015). Haven't dug into it in detail, on my phone and no longer at Adafruit, but I think it's a case of Nordic moving to a new 5 clause license after the fact, which they're perfectly entitled to do.

dhalbert, 2023-09-30:
The license appears to have been changed to the 5-clause license here: https://github.com/NordicSemiconductor/pc-nrfutil/commit/48b618967f334743b5937d1283fbf8837c4337e7 (Jun 21, 2016). The original Python code is no longer supported by Nordic: they have a downloadable compiled tool also known as `nrfutil`.

As mentioned in https://pypi.org/project/adafruit-nrfutil/, the license is "other". Given that it's available in PyPi, I'd say leave it as that, and suggest you not package it.

I will look at the license settings in this repo and correct them if needed.

pehjota, 2023-09-30:
> It looks to me in the first link in your comment ("use") that Nordic itself changed their license in 2016 in that commit from Carles, and we made our fork before those changes (2015). Haven't dug [...]

Thanks, you're right, Nordic added clauses 4 and 5, not the other way around (sorry for suggesting as much and for the confusion).  Although the commit history says this fork is from 2017, after Nordic's license change, but you probably forked a version from before the change.

So it looks like the license on this project is accurate, except nordicsemi/dfu/signing.py seems to refer to the non-free 5-clause license.

@carlescufi Would Nordic be willing to relicense [nordicsemi/dfu/signing.py](https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/nordicsemi/dfu/signing.py) (or even the entire old nrfutil code) under the [previous](https://github.com/NordicSemiconductor/pc-nrfutil/commit/48b618967f334743b5937d1283fbf8837c4337e7) BSD 3-clause [license](https://github.com/NordicSemiconductor/pc-nrfutil/blob/6698b6d91577527198ab0ced5152ff3279c15925/license.txt)?  I'd open an issue over there, but the project is archived.

carlescufi, 2023-10-21:
> @carlescufi Would Nordic be willing to relicense [nordicsemi/dfu/signing.py](https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/nordicsemi/dfu/signing.py) (or even the entire old nrfutil [...]

@pehjota Maybe, but what is the issue with the Nordic 5-clause license? If you are going to use this with Nordic products does it impact you?

emilazy, 2024-07-14:
@carlescufi Use restrictions like that make the licence not qualify as Free/open source software, so it – and any software that depends on it – can’t be packaged in many Linux distributions (e.g. Debian and Fedora). I don’t think the licence for some DFU glue code provides any meaningful commercial protection to Nordic, so if it is possible to get it relicensed under an FSF/OSI‐approved FOSS licence I would highly recommend doing so, and I am sure the Nordic user community would be grateful for the gesture!