adafruit/Adafruit_nRF52_nrfutil #45: Failed to import ecdsa, cannot do signing

jamcinnes2, 2024-11-21:
### Operating System

Linux

### IDE version

Arduino 2.3.3

### Board

nRF52

### Version

latest

### What happened ?

I "compiled" adafruit-nrfutil from source and created the stand alone self contained binary. Everytime I run it I get this message in the console: "Failed to import ecdsa, cannot do signing"

But it still works, I can upload my code to my board.

### How to reproduce ?

1. clone this repo
2. install pip requirements
3. create the self contained binary & run it

dhalbert, 2024-11-21:
How did you install the requirements? Did you do
```sh
pip install -r requirements.txt
```
`requirements.txt` includes `ecdsa`.

jamcinnes2, 2024-11-26:
I did do that command and install ecdsa.

I think I found my problem. I tried to create the self containted executable on Linux. Maybe this is only supported on Windows and MacOS? If I do the normal `python setup.py install` then I get a working command with no error message.

closed 2024-11-26 by jamcinnes2